erdemarslan/GSMSim #4: Can't gsm.gprsHTTPGet to deal with long URLs

gusantor, 2018-07-26:
Hello, thanks for your library

Using gsm.gprsHTTPGet , I'm able to get results from URLs like www.google.com or www.mysite.com

But not from www.mysite.com/one/or/two/branches

It's just me ?

Thanks in advance

gusantor, 2018-07-26:
Hello

I've just tried, same results ... sort of intermittent, never get a 200 response code, except when I use just a domain (www.google.com, www.mysite.com)

Never the less, I always get an IP

Response I wait for is just json 180 chars

..and I needed to add #include <Wire.h>, if not, arduino UNO keeps rebooting

Thanks

erdemarslan, 2018-08-01:
Can you send your codes and connection diagram to me? I want to test it.

closed 2020-05-01 by erdemarslan

erdemarslan, 2020-05-07:
GSMSim is updated. Please check the examples. You can update library from Arduino Library Manager or manually.