erdemarslan/GSMSim #8: smsREAD not working

zsarwar, 2018-11-10:
Hi,

I have been trying to use the SMSread function. 
It always returns back index no error, however manual executing the +CMGR is working fine.
Is there any update

erdemarslan, 2018-11-12:
Can you send your codes and any screenshot for this.

zsarwar, 2018-11-18:
void loop() {
// put your main code here, to run repeatedly:

if((gsm.smsListUnread())=="NO_SMS")
{
 Serial.println("Awaiting new SMS");
 Serial.println("");
 delay(1000);
}
else
{
 Serial.println("New SMS Recevied");
 Serial.println("");
 delay(1000);

 Serial.print("Reading SMS : ");
 Serial.println(gsm.smsRead(1));
 Serial.println("");
 delay(1000);  
 
 for(;;) {} // This test code will put endless loop and stop furtehr execution

}

[image]

erdemarslan, 2018-11-18:
Your error about message index no error. There is no message on index no = 1 in your sim card or module.

Change your codes like this in your loop.

`
void loop() {
  String indexno = gsm.smsListUnread();

  if indexno == "NO_SMS" {
    Serial.println("Awaiting new SMS");
    Serial.println("");
    delay(1000);
  } else {
    Serial.println("New SMS Recevied");
    Serial.println("");
    delay(1000);
    // now indexno like this => 21 or 23,24 etc
    // check is it contains comma (,)
    if indexno.indexOf(",") == -1 {
      // only one number
      Serial.print("Reading SMS : ");
      Serial.println(gsm.smsRead(atoi(indexno)));
      Serial.println("");
      delay(1000);
    } else {
      // split each number whichs are seperated with comma
      // then read each message in loop
      // You can look this example for this
      // https://arduino.stackexchange.com/questions/1013/how-do-i-split-an-incoming-string/1033#1033
    }
  }
}
`

closed 2018-11-18 by erdemarslan

zsarwar, 2018-11-18:
Still getting error while compiling.

[image]

erdemarslan, 2018-11-18:
Change marked line to

`Serial.println(gsm.smsRead(indexno.toInt()));`

zsarwar, 2018-11-18:
Getting the same Index no error message again 

[image]

erdemarslan, 2018-11-18:
Can you print indexno variable? 
Please add
`Serial.println(indexno);`
and add your screenshot here.

zsarwar, 2018-11-18:
Sent

zsarwar, 2018-11-24:
Can you please confirm if you have received my code

erdemarslan, 2018-11-26:
Yes your code received to me. I test it and there is no error. This issue about incoming message indicator. You must change incoming message indicator. You must change your message indicator to save option. I missed it when i coding this library. I update codes and add smsChangeIncomingIndicator function. You can change your indicator to show or save. smsChangeIncomingIndicator(true) save message for read. smsChangeIncomingIndicator(false) show the messages while program runnig.

I advise dont list messages, read them when they come. So i added readSerial command for this.

zsarwar, 2018-11-28:
Dear ,
smsRead() is still not working. 
I have simplified my code as below and the output as followed.

[image]

[image]

erdemarslan, 2020-05-07:
GSMSim is updated. Please check the examples. You can update library from Arduino Library Manager or manually.