erdemarslan/GSMSim #9: No method to get the senders number

tauki, 2018-11-24:
If we look into the SIM800L library by cristiansteib, there's a method that retrieves the SMS senders number. Which is a pretty important feature. 
Following is the code segment from cristiansteib's library.

```
String Sim800l::getNumberSms(uint8_t index){
  _buffer=readSms(index);
  Serial.println(_buffer.length());
  if (_buffer.length() > 10) //avoid empty sms
  {
    uint8_t _idx1=_buffer.indexOf("+CMGR:");
    _idx1=_buffer.indexOf("\",\"",_idx1+1);
    return _buffer.substring(_idx1+3,_buffer.indexOf("\",\"",_idx1+4));
  }else{
    return "";
  }
}
```

erdemarslan, 2020-05-02:
In new version i added a function for this. Thanks.

closed 2020-05-02 by erdemarslan

erdemarslan, 2020-05-07:
GSMSim is updated. Please check the examples. You can update library from Arduino Library Manager or manually.