erdemarslan/GSMSim #14: Manually send AT commands

progressify, 2019-05-24:
Hi, 

I have tested some SIM800L EVB module (with blue PCB), if I use the library not work correctly, because it does not catch the signal by the carrier. 

But if I use the AT command for manually catching the signal:

```
AT+COPS=?
+COPS: (2,"vodafone","voda IT","22210"),(1,"Wind Telecom SpA","I WIND","22288"),,(0-4),(0-2)
```
for listing all cell operator available and I manually set the operator:
```
AT+COPS=0,0,"I WIND"
OK
```
after this operations, I can send correctly SMS, or a call number, etc...

In your library exist some method for send manually `AT+COPS` commands?

erdemarslan, 2020-05-01:
I will release new version in a few days. This function will be add. Thanks.

closed 2020-05-01 by erdemarslan

erdemarslan, 2020-05-07:
GSMSim is updated. Please check the examples. You can update library from Arduino Library Manager or manually.