erdemarslan/GSMSim #15: Gmail function not working

amansinghaljpr, 2019-05-24:
I have tried using the gmail function but i think there is a problem with the "AT+SMTPFROM" part because when i enter the correct email in the .ino example even then it quits out the loop and tells me : 
GSMSim Library - Send Email Example

Connect to Internet: 0
Set gmail smtp configuration: 0
Write email to someone: ERROR:FROM_NOT_SET
Send email... : ERROR:EMAIL_SENDING_ERROR
Close Internet: 0

erdemarslan, 2019-05-24:
You must make some settings on GMail. Are you look this information on https://github.com/erdemarslan/GSMSim/blob/master/examples/GSMSim_Email/GSMSim_Email.ino

// Set Gmail Account
  // If you take an error when sending email with google, please check display unlock captcha setting on google
  // please use this url for this.
  // https://www.google.com/accounts/DisplayUnlockCaptcha
  // for gmail, dont use @gmail.com on your username.
// at the last, if you take an authendication error, please check 2 way verification settings on your account.

amansinghaljpr, 2019-05-25:
I have disabled the capcha and there is no 2 step verification on my account but even then i am getting the same problem. I am entering the username without @gmail.com where it needs to authenticate and login into my account. But i guess problem occurs when the code steps onto the "from email" and "to email" part there it exits out of the loop and simply throws an error just like i showed you . I guess probable solution can be you giving me a rough example code with false emails and false sender and receiver just to give me an idea that am i doing this right or not .

One more question i had was that have you ever tried this code yourself or have you tested this code before on a sim800l module before ?

erdemarslan, 2020-05-02:
> I have disabled the capcha and there is no 2 step verification on my account but even then i am getting the same problem. I am entering the username without @gmail.com where it needs to authenticate [...]

Yes i will try it. I will check it again. Now i am working on new version. I look this double.

closed 2020-05-02 by erdemarslan

erdemarslan, 2020-05-07:
GSMSim is updated. Please check the examples. You can update library from Arduino Library Manager or manually.