erdemarslan/GSMSim #20: dtmf

atulgo, 2019-12-30:
Detection of DTMF tones - SIM800L module.

I get "OK" when I send 'AT+DDET=1,1000,1,1* ' over uart .
But it doesnt seem to detect anything afterwards.
Perhaps I am doing something wrong.
Would appreciate some feedback.

regards
Atul
ref : *SIM800 Series_AT Command Manual_V1.09

erdemarslan, 2019-12-30:
Hi,
When you send at command to module, if your command is true and understand
by module , it returns OK. At dtmf command module ask to answer operator
and oparator turn answer to module. Than module say to answer to you. So
you must listen module for answer for a while. Check your dtmf command. May
it not be understandable for operator. Check your module return messages
show options.

May your code wrong.

30 Ara 2019 Pzt, saat 14:42 tarihinde atul <notifications@github.com> şunu
yazdı:

> Detection of DTMF tones - SIM800L module.
>
> I get "OK" when I send 'AT+DDET=1,1000,1,1* ' over uart .
> But it doesnt seem to detect anything afterwards.
> Perhaps I am doing something wrong.
> Would appreciate some feedback.
>
> regards
> Atul
> ref : *SIM800 Series_AT Command Manual_V1.09
-- 
iPhone ile Gmail Mobil üzerinden gönderildi.

atulgo, 2020-01-04:
Thanks for the reply Eredm.
I was testing with a serial communication prog called minicom... so code doesnt come into picture.
but it does seem an issue at operator end.
Closing.

closed 2020-01-04 by atulgo

erdemarslan, 2020-05-07:
GSMSim is updated. Please check the examples. You can update library from Arduino Library Manager or manually.