erdemarslan/GSMSim #28: HttpPost with uint8_t*

rtarta, 2020-05-08:
Hi we need to post uint8_t* instead of String without convetion.

erdemarslan, 2020-05-08:
Hi can you explain it more?

rtarta, 2020-05-13:
Hi @erdemarslan  On your post sample we need to post file or a little big data. Could you please check the  below address.  
https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.h#L165

HACKER-3000, 2020-07-03:
Maybe check this: https://github.com/erdemarslan/GSMSim/blob/master/examples/GSMSim_HTTP/GSMSim_HTTP.ino#L110

jaya-shankar, 2020-08-31:
hey, Guys, I am new to this I have a doubt in the given documentation getWithSSL(String URL) and postWithSSL(String url, String data, String contentType)  are nothing but **HTTPS (secure)** requests right??

erdemarslan, 2021-08-24:
> hey, Guys, I am new to this I have a doubt in the given documentation getWithSSL(String URL) and postWithSSL(String url, String data, String contentType) are nothing but **HTTPS (secure)** requests right??

Yes