erdemarslan/GSMSim #32: gprsIsConnected() always return true

AndiP1234, 2020-06-25:
I have some problems with gprsIsConnected at v1.xx.
After the connection request, I have some bad characters at the beginning of the _buffer like:
Send: AT+SAPBR=2,1
Answer: ⸮+ATSAPBR.....
indexOf returns always -1, I think there is an issue at indexOf()..

Now I debugged the _buffer and the first char is 0xE0 = 224 = à

if (_buffer.indexOf("ERR") != -1 || _buffer.indexOf(""0.0.0.0"") != -1 **|| _buffer.substring(4).indexOf(""0.0.0.0"") != -1**) {

AndiP1234, 2020-06-25:
other solution, cleaup the _buffer:
```ruby
void GSMSim::_readSerial(uint32_t timeout) {
	_buffer = "";
	uint64_t timeOld = millis();
	// Bir süre bekle...
	while (!gsm.available() && !(millis() > timeOld + timeout)) { ; }
	// beklemeden çıkınca ya da süre dolunca varsa seriali oku, yoksa çık git...
	if(gsm.available()) { _buffer = gsm.readString(); }
	char check[10];
	_buffer.toCharArray(check, 10);
	if (check[0] == 0xE0){
		_buffer = _buffer.substring(1);
	}
	_buffer.trim();
}
```

@erdemarslan: can you explain why you use for many functions more than one? e.g. _readSerial() and _readSerial(uint32_t timeout). You can cleanup your code if you change the header file:
_readSerial(uint32_t timeout = TIME_OUT_READ_SERIAL);
With this you only need the readSerial function only one time at *.cpp

HACKER-3000, 2020-07-03:
That could be the reason for my weather station constatly crashing... It always got stuck on connecting until i added the Watchdog Timer.

closed 2021-08-24 by erdemarslan