erdemarslan/GSMSim #35: library not working with sim7020e

roysG, 2021-03-24:
Hi,
I have module sim7020e, i tried to test it with library called Adafruit FONA Library and it found the card and show me his info with no problem!

My next step was to try to work with you library, but the connection is in initial and your library cant find the modem device.

Please help me, what may be the problem??
Thanks.

roysG, 2021-03-24:
The connected pins in the sim7020e:

sim7020e  -> arduino nano
---------------------
VIN             -> 5V
GROUND   -> GROUND
TX              -> D4
RX              -> D5

(I cant find the reset pin, does it important?)

erdemarslan, 2021-04-14:
Hi, reset pin is not important. Some modules hasnt got reset pin. You can leave it blank or give fake pin. I dont have sim7000 series module yet. So i cant test it. If your device not communicate to your arduino, please check RX and TX pins. UART communication is same for all modules.

Sim800 and Sim900 series are GSM based modules with 2G. If i dont remember wrong, Sim5000 series are 3G and Sim7000 series are 4G modules. May some AT commands are not same with Sim800 and 900 series. I must look AT commands datasheet for this. When i wrote this library i had tested all commands with my Sim800L module. I can add Sim7000 or Sim5000 series AT commands to library but i dont have this modules and i cant test them. This is a huge work for me because i must compare all AT commands of Sim800, Sim5000 and Sim7000 series. May be in future.

closed 2021-08-24 by erdemarslan