erdemarslan/GSMSim #36: Is this still being worked on ?

TravisDuck999, 2021-04-14:
Hi, just wondering if this library is still being worked on ?

I would love to use this with the SIM7000G  ( currently using the LiliyGo SIM7000G board ) but it crashes the board when I load it.
This might be something to do with the board but I thought I would ask if the SIM7000 was supported.

Thanks for your time and great work.

erdemarslan, 2021-04-14:
Hi, I am a teacher and we are currently in the pandemic process. Therefore, our lessons are held online. It takes a lot of time to prepare for online classes, prepare presentations, videos and experiments, assign homework and projects and check them out. Maybe I can make a detailed review when we go on summer vacation. 

Also, I do not have a SIM7000 series module. Therefore, I do not have the opportunity to test the codes I wrote. This library was written using AT commands of SIM800 and SIM900 series modules. While reworking the library, I have to compare the AT command datasheets of the SIM800 and SIM900, SIM5000 and SIM7000 series. This will take quite a while. Of course, first of all I have to buy a SIM7000 series module. :)

In the description of the library, I stated that it was tested on SIM800L and it may not work on other modules or some of them may work. ->"This library tested on Sim800L (5V Module). Some methods may not work all SimCom Modules like FM Radio methods. This library may work other gsm shields or modules. But some AT Commands only work SimCom modules. So some functions may not work on other gsm shield or module (etc A6 Ai-Thinker)"

If you report the problem by doing a detailed debug, we may come up with a solution in the short term. Thank you for your understanding.

TravisDuck999, 2021-04-15:
Hi, I completly understand.

Thank you so much for your time and commitment, I'll have a look back every now and then.

Until then, good luck with everything and thank you again for all the amazing work.

closed 2021-04-15 by TravisDuck999