erdemarslan/GSMSim #37: questions: How To use string as sms message?

Trpplayer79, 2021-05-16:
How can you use a string variable as sms message?

For example:

```
#define SMSNUMBER "+3xxxxxxxxxx"
String MESSAGE = "";   // depending on the reason is this filled with different text by the program/sketch/code

void loop() {
sms.send(SMSNUMBER, MESSAGE);
}
```
Tthis gives the fault message:

```
In function 'void SMSSend()': [Sketch]:[Line:Char]: error: no matching function for call to 
'GSMSimSMS::send(const char [13], String&)' sms.send(SMSNUMBER, MESSAGE);
```

When i remove the variable as message does the code compile correctly:
```
sms.send(SMSNUMBER, "Test Message");
```

erdemarslan, 2021-05-16:
Please try to convert string to char array. Like this.
`String MESSAGE="This is message";
int strLen = MESSAGE.lenght() + 1;
char messageChar[strLen];
MESSAGE.toCharArray(messageChar, strLen);
sms.send(SMSNUMBER, messageChar);
`

erdemarslan, 2021-05-16:
If your issue solved, please close topic. Thanks.

Trpplayer79, 2021-05-17:
Thank you, This solved the issue!

Suggestion: integrate this somehow into the example sketch.

closed 2021-05-17 by Trpplayer79