erdemarslan/GSMSim #47: Wrong return value in send method

peacekiller, 2022-10-05:
Hi Erdem,

first let me thank you for developing this lib. I want to use it to realize my sms gateway using the SIM800L on an ESP8266 arduino board.

I get a compile error on this line:

https://github.com/erdemarslan/GSMSim/blob/337a9c89063a28349cff6cfc7c4602c5db3dfad6/src/GSMSimSMS.cpp#L148

The defined return type of the send method is `boolean`. But you try to return a string. It looks like a unwanted copypaste...

best regards

Slawwoj, 2023-01-29:
Were you able to resolve this problem? Because i have the same issue and any help would be great

jaroboch, 2025-01-26:
bool GSMSimSMS::send(char* number, char* message) {

	gsm.print(F("AT+CMGS=\""));  // command to send sms
	gsm.print(number);
	gsm.print(F("\"\r"));
	_readSerial();  //wait for  '>'
	vTaskDelay(100 / portTICK_PERIOD_MS);	//vTaskDelay dla FreeRTOS
	//delay(100);
	gsm.print(message);
	gsm.print("\r");
	vTaskDelay(100 / portTICK_PERIOD_MS); //vTaskDelay dla FreeRTOS	
	//delay(100);
	gsm.print((char)26);
	vTaskDelay(100 / portTICK_PERIOD_MS); //vTaskDelay dla FreeRTOS	
	//delay(100);
	_readSerial(20000);

	if (_buffer.indexOf("+CMGS:") != -1) {
		return true;
	}
	else {
		return false;
	}
}